eigger/espcomponents #61: ESP HOME을 이용한 코콤 월패드 제어에 대한 보다 자세한 설명을 얻을 수 있을까요?

chodadoo, 2025-02-11:
ESP HOME을 이용한 코콤 월패드 제어에 대한 보다 자세한 설명을 얻을 수 있을까요?

무지성으로 완전 무장된 상태라 그런지
https://github.com/eigger/espcomponents/blob/master/packages/wallpad/README.md 만으로는
어떻게 시작을 해야 할지 모르겠네요 ...

eigger, 2025-02-11:
편하게 물어보셔도 되요.
esphome을 사용하셔서 펌웨어를 올려 보셨을까요?
아니면 ew11을 해보셨을까요?

eigger, 2025-02-11:
https://github.com/eigger/espcomponents/wiki/ESPHome-시작하기-(KOCOM)
여기 한번 봐보세요. 그림은 추가 못했네요

chodadoo, 2025-02-12:
> 편하게 물어보셔도 되요. esphome을 사용하셔서 펌웨어를 올려 보셨을까요? 아니면 ew11을 해보셨을까요?

감사합니다.
EW11 을 사용중에 있습니다. ESP 롬라이팅 경험이 없어 댓글 주신 내용을 진행해도록 하겠습니다.
친절한 답변 너무 감사합니다.

eigger, 2025-02-12:
근데 저도 코콤 입주예정이고 입주를 아직 안해서 코콤 현관문만 사용중이고 코콤 월패드는 아직 실사용을 안해봤어요.
그래서 패킷 검증이랑 구성요소 추가를 좀더 해야 할수도 있어요.
제가 입주하고 검증되면 추가로 업데이트 할거긴 한데 잘 안되시면 댓글 남겨주세요

closed 2025-02-13 by eigger