eigger/espcomponents #71: Would you like me to write some documentation?

matwho, 2025-03-09:
This is a very useful external component - it really fills a gap is ESPHome that has caused me lots of unnecessary work/hacks to parse UART data.

However, I think it needs more documentation. The examples you give helps but is missing the overall flow of data through this component. It will be very hard for someone to pick this up from the documentation there is at the moment.

I would like to have a go at writing a guide? Would you help correct the bits I have misunderstood about your code?

Thanks for this very useful software!!

eigger, 2025-03-09:
Thank you for your interest. Actually, I originally made this for my personal use, so the documentation is lacking. I'm happy to help with anything I can.

matwho, 2025-03-09:
Oh great I will start writing! Your component is realy going to be very popular!!

matwho, 2025-03-14:
I had a bit of spare time today so made a start on adding documentation: your early comments would be appreciated- I have put :shrug: on bits I would like some help understanding!

Have a look here: https://github.com/matwho/espcomponents/tree/dev/components/uartex

eigger, 2025-03-15:
Sorry, I was only planning to write the documentation but never actually got around to doing it. 
Your documentation explains the component in detail, making it easy for first-time users to understand. It's great!
I will provide additional explanations.
 
1. uint8_t = (uint8_t* data, uint16_t len) 🤷    
    Everything written below the lambda represents the parameter types, variable names, and the return type.
    uint8_t* data, uint16_t len This value receives only the actual data, excluding the header and footer checksums.

2. rx_checksum2, tx_checksum2
    This is a setting for a checksum that uses 2 bytes, and when using a lambda, the checksum can be set to n bytes."

3. state configuration
    The state is used to filter and verify whether the status value in a common packet parsed by Uartex matches the corresponding device. In RS485 or similar networks with multiple connected devices, it helps each device determine if the status value belongs to itself
    When a packet arrives, it is first parsed by Uartex and then distributed to all Uartex entities. Each entity then parses the received packet using state to verify whether it belongs to that entity. If the packet matches, the entity proceeds with a status check, such as turning on or off.

    > data: The packet to be checked
    > mask: Mask function
    > offset: The position to start checking
    > inverted: Whether the packet matches or does not match

4. precision
    If the value obtained from the packet is 123, then:
    > With precision = 0, the value remains 123.
    > With precision = 1, the value becomes 12.3.
    > With precision = 2, the value becomes 1.23.

matwho, 2025-03-15:
It’s a pleasure to help out - I think it is always better if someone who did not write the code writes the documentation. They can see it from the same angle as a first time user more easily.

I will keep asking questions - if I put your answers in the documentation in the long term it will reduce the number of times you get asked. I hope!

This feature has been missing from ESPhome for a long time - it will be very popular!

eigger, 2025-03-15:
Feel free to ask anytime! I believe that collaborating on the documentation will help the component evolve in a better direction.

matwho, 2025-03-18:
As i am writing documentation I have some more questions and things I would like to check with you!

1) With rx_checksum2 exactly how is the CRC calculated and checked - I am guessing you calculate the CRC after every byte is received and then check that byte and the next one in the data - is that correct?

2) Just checking tx_timeout is the time the component waits before assuming an ACK has not been sent by the device and then it tries to send command again for tx_retry_cnt number of times before giving up.

3) What do you do to the GPIO pin specified in tx_ctrl_pin, does it indicate when data is being sent, if so exactly how?

4) Just checking x_footer adds the data after the CRC i.e. the footer is the very last thing sent.

5) I think we should include some clear code/psudo-code fragments for the algorithm used to calculate CRC with `add` and `xor`

6) Just checking, in the `state:` section does `inverted: true` mean the `data:` must not match to be processed or us the data inverted before a comparison is made?

7) What is the default when there is no `state:` section? What is the easiest way to specify "match on every packet"?

More to come!

eigger, 2025-03-18:
1. The CRC calculation using add and xor appends a 2-byte CRC value to the data.
2. Yes, that's correct.
3. Certain RS485 converters require controlling a GPIO pin, which must be set to High when transmitting data.
4. Yes, that's correct.
5.
```
    uint16_t crc = 0;
    switch(checksum)
    {
    case CHECKSUM_XOR:
        for (uint8_t byte : header) { crc ^= byte; }
        for (uint8_t byte : data) { crc ^= byte; }
        break;
    case CHECKSUM_ADD:
        for (uint8_t byte : header) { crc += byte; }
        for (uint8_t byte : data) { crc += byte; }
        break;
    case CHECKSUM_XOR_NO_HEADER:
        for (uint8_t byte : data) { crc ^= byte; }
        break;
    case CHECKSUM_ADD_NO_HEADER:
        for (uint8_t byte : data) { crc += byte; }
        break;
    }
    return crc;
```
checksum: crc & 0xFF
checksum2: crc >> 8, crc &0xFF
6. The result of the data comparison is inverted. In other words, the data must not match for it to be processed.
7. When the state: section is absent, it matches all packets. The easiest way to match all packets is to remove the state: section.

Let me know if you have any other questions.

matwho, 2025-03-24:
Thanks for all the answers it is getting much clearer now!

Here is one more question:

1. In your psudo-code showing how the CRC is calculated you have a version including the header and one without (“_NO_HEADER”). With the YAML configuration how do we specify which one we want to use.

eigger, 2025-03-24:
It's simply one of the checksum types.
```
rx_checksum: add
rx_checksum: xor
rx_checksum: add_no_header
rx_checksum: xor_no_header
```

closed 2025-04-26 by eigger