eigger/espcomponents #72: Header bytes missing from crc calculation

matwho, 2025-03-09:
This is a better place to discuss this issue.

The CRC check is not working for me. I put this in for the CRC calculation:

uartex:
  rx_timeout: 150ms
  tx_delay: 10ms
  tx_timeout: 10ms
  tx_retry_cnt: 3
  rx_header: [0xCA, 0xFD, 0x06, 0x2F, 0x90]
  rx_checksum: !lambda |-
    uint32_t crc = 0;
    for (int i = 0; i < len; i++)
    {
      crc += data[i];
      ESP_LOGD("CRC","%#d %#x",i,data[i]);
    }
    ESP_LOGD("CRC END","%#x",crc);
    return crc & 0x00FF;

Just to see what was going on!

I need to include the header bytes in the CRC calculation - but you have removed them before they get to the lambda. I only get the bytes after the header. I guess the same is true when you do the calculation with rx_checksum: add.

I imagine we need an option to choose what is included in the CRC calculation.

Something like this, as an alternative to rx_checksum: add

  rx_checksum:
    include_header: true
    type: add

eigger, 2025-03-10:
The current rx_checksum: add option includes the header. Could you test it using a lambda function? If you provide the raw packet, I can check it as well.
```
  rx_checksum: !lambda |-
    uint32_t crc = 0xCA + 0xFD + 0x06 + 0x2F + 0x90;
    for (int i = 0; i < len; i++)
    {
      crc += data[i];
      ESP_LOGD("CRC","%#d %#x",i,data[i]);
    }
    ESP_LOGD("CRC END","%#x",crc);
    return crc & 0x00FF;
```

eigger, 2025-03-10:
Your checksum is indeed an add checksum. Does setting rx_checksum: add cause it to malfunction?
If it does not function correctly, could you provide the logs?

matwho, 2025-03-10:
This is with `rx_checksum: add` and `rx_timeout: 150ms` 

You can see I get Component uartex took a long time for an operation warning after every block.

[logs_trickle-charger_logs.txt](https://github.com/user-attachments/files/19166656/logs_trickle-charger_logs.txt)

eigger, 2025-03-11:
A long-time warning occurring does not mean that the checksum is not working. The current add checksum is functioning normally. If a timeout or checksum error occurs, it will be displayed as an error on the error sensor rather than as a warning log. The exact cause of the long-time warning is not known at this time, but it could be due to the time taken to output debug logs or the presence of many uartex sensors. Could you try changing the debug level to DEBUG and check if the warning still occurs?

matwho, 2025-03-11:
Yes you are correct logging above `WARN` causes blocking for too long. I removed the `VERY_VERBOSE `  replaced it with 'WARN` only get one warning now first time through after boot up as something is unknown.

Attached is the log (but with logging set to `DEBUG` to give your more information - so ignore all warnings about blocking for too long, these don't happen at `WARN` level)

[logs_trickle-charger_run (3).txt](https://github.com/user-attachments/files/19181722/logs_trickle-charger_run.3.txt)

matwho, 2025-03-12:
It is a pleasure to help - thanks for creating such a useful component!

closed 2025-03-12 by matwho