eigger/espcomponents #74: 저희집에서 정상 작동하지 않는 것 같습니다.

chodadoo, 2025-03-11:
기존 EW11 사용하였으며
애드온 'https://github.com/clipman/kocom.py' 이용중에 있었습니다.

금일 설명서 만들어 주신 것 보고 펌웨어 올리고 설정 후 연결까지 모두 진행하였습니다.

그런데
방 0(첫번째) / 라이트 0(첫번째) 켜고, 끄기 기준... 작동하지 않네요
혹시 'kocom.yaml'을 각 가정에 맞게 수정하여 쓰는게 기본일까요?

기존 사용하던 애드온과 패킷을 비교하니 그 값이 좀 다르기는 합니다.
추가로 ESP 이용시 RECV 패킷이 로그에 나오지 않는데 정상인가요? 아니면 RECV를 못 받고 있는걸까요? ...

기존 애드폰 패킷 로그
INFO[2025-03-11 19:53:32,372]:[ACK] OK 
INFO[2025-03-11 19:53:32,767]:[recv] aa5530bc0001000e000000ff000000000000fa0d0d 
INFO[2025-03-11 19:53:32,769]:[recv] aa5530dc000e0001000000ff0000000000001a0d0d 
INFO[2025-03-11 19:53:32,771]:[MQTT publish|light] data[{'light_1': 'off', 'light_2': 'on', 'light_3': 'off'}] 
INFO[2025-03-11 19:53:34,594]:[MQTT RECV] kocom/livingroom/light/1/command 0 b'on' 
INFO[2025-03-11 19:53:34,597]:[cache|light0] query cache 30dc000e0001000000ff000000000000 
INFO[2025-03-11 19:53:34,601]:[SEND|light] aa5530bc000e00010000ffff000000000000f90d0d 
INFO[2025-03-11 19:53:34,721]:[recv] aa5530dc0001000e0000ffff000000000000190d0d 
INFO[2025-03-11 19:53:34,723]:[ACK] OK 
INFO[2025-03-11 19:53:35,118]:[recv] aa5530bc0001000e0000ffff000000000000f90d0d 
INFO[2025-03-11 19:53:35,121]:[recv] aa5530dc000e00010000ffff000000000000190d0d 
INFO[2025-03-11 19:53:35,123]:[MQTT publish|light] data[{'light_1': 'on', 'light_2': 'on', 'light_3': 'off'}] 

ESP HOME 로그
ESP Kocom Log [W]AA5530BC000E000100000000000000000000FA0D0D(21) (2)로 변경되었습니다.
오후 7:55:36 - 15분 전
ESP Kocom Log [W]AA5530BC000E000100000000000000000000FA0D0D(21) (1)로 변경되었습니다.
오후 7:55:35 - 15분 전
ESP Kocom Log [W]AA5530BC000E000100000000000000000000FA0D0D(21)로 변경되었습니다.

eigger, 2025-03-11:
Recv 패킷을 받아야 하는데 못받고 있는거 같아요. 그러면 제대로 패킷이 전송이 안될걸수도 있어요
kocom.yaml도 각 가정에 따라 수정하긴 해야합니다.

chodadoo, 2025-03-12:
동일한 배선이며 잘 못되었나 싶어 교차 배선도 해보고 테스트 했습니다.

kocom.yaml 을 수정하여 사용할 경우
esp 빌더에서 esp-kocom.yaml 의 패키지 내용 중 kocom.yaml을 지우고 그 내용을 하단에 붙여도 작동할까요???
찾아보니 레파지터리를 포크하고 사용 하시는것 같던데 ... 질문드린 내용으로 처리해도 괜찮은지 궁금하네요

eigger, 2025-03-12:
패킷이 달라서 확인해보니 체크섬이 잘못 되어 있었네요
저도 아직 실사용을 안해봐서 검증을 못해서..

다시 펌웨어 올려 보시겠어요? 

한번 펌웨어가 올라간 상태니 USB 연결 필요 없고
Install - ✅Wirelessly 클릭 하시면 다시 펌웨어 업로드 됩니다.

그리고 저는 관리 및 공유 목적으로 깃허브에 올려서 사용 하는거라 테스트 및 개인적인 사용 목적이면
esphome에 그대로 넣고 쓰셔도 됩니다.

```
packages:
  remote:
    refresh: always
    url: https://github.com/eigger/espcomponents/
    files:
      - packages/device_base.yaml
      - packages/esp32.yaml
      # - packages/wallpad/kocom/kocom.yaml

uart:
  baud_rate: 9600

....

```
이런식으로  packages/wallpad/kocom/kocom.yaml 을 지우거나 주석 처리 하시고 그 안에 내용을 밑에 붙여 넣기 하셔서 사용 하시면 됩니다.

지금 에드온에서 사용하시는 제어 요소를 알려주시면 구성하는데 도움을 드릴께요

chodadoo, 2025-03-12:
다시 펌웨어 올리니 정상 작동합니다 ! 감사합니다.

방수 조명수가 달라 조정을 해야하는데,
대체로 포크해서 사용하는지 esp-kocom.yaml 에 수정을 하여 사용하는지 궁금합니다. (대세론을 찬양)

전열교환기가 누락되어있는데 직접 한번 해봐야겠습니다.

closed 2025-03-12 by chodadoo

eigger, 2025-03-12:
에드온은 포크 해서 개인이 설정을 따로 해야 되는 부분이고 esphome은 설정이 따로 가능해서 굳이 포크 안하셔도 됩니다.