eigger/espcomponents #77: Feature request : on_timeout

matwho, 2025-03-16:
I have a device that goes to low power mode after a time. It stops responding to its UART, I then have to send 2s of zeroes which wakes it up.

I think that at the moment with your component I have to just assume it’s asleep and send the wake up each time. It would be nice to have:

```
command:
  …
  on_tx_timeout:
    then:
      - script.execute: …
```

Then I can wake it up when I need. This would be generally very useful. Might be useful for `rx_timeout` and crc errors as well.

Thanks.

eigger, 2025-03-16:
Implementation is feasible, but i need to consider the operation in more detail. Currently, tx_timeout only functions for commands with ack enabled, meaning it represents the waiting time until an ack is received. When a timeout occurs, uartex will retry up to the specified retry_cnt, with the timeout applied to each retry attempt. I need to decide whether the on_timeout event should trigger every time a timeout occurs or only once after exceeding the retry count.

matwho, 2025-03-16:
I think after `retry_cnt` had been exhausted and basically the tx has completely failed would be best, this is when “something else” needs to be done.

eigger, 2025-03-17:
I have made several modifications and added both the new functionality and the ack mask feature. For tx_timeout to occur, ack must be enabled, and because your packet format places the value immediately after the header, it is not possible to verify a specific packet for ack. Therefore, mask processing was necessary

```
uartex:
  on_tx_timeout:
    then:
      - logger.log: "Timeout error"

button:
  - platform: uartex
    name: "Test"
    command_on: 
      data: [0xCA, 0xFD, 0x00, 0x06, 0x09, 0xD6]
      ack: [0x00]
      mask: [0x00]
```

matwho, 2025-03-18:
Great! I will test this and also include in documentation.

closed 2025-03-20 by eigger