eigger/espcomponents #146: [W][uartex:332]: [Read] Size error:

JSa1987, 2025-05-31:
Hello,

First of all thank you for creating the uartx component! This seems to be extremely useful. 
I'm trying use this component for the first time and I'm having some issue reading the UART data. I have setup a Media Player entity and the commands work fine. However, seems that the component is unable to read any of the UART data. Each time the player sends an UART message I get following "Size error":

```
[21:59:13][W][uartex:332]: [Read] Size error: 9800(2)
[21:59:13][D][uart_debug:114]: <<< 98:00:0A
[21:59:16][W][uartex:332]: [Read] Size error: 9850(2)
[21:59:16][D][uart_debug:114]: <<< 98:50:01:01:04:53:0A
[21:59:22][W][uartex:332]: [Read] Size error: 98(1)
[21:59:22][D][uart_debug:114]: <<< 98:01:0A
[21:59:30][W][uartex:332]: [Read] Size error: 9800(2)
[21:59:30][D][uart_debug:114]: <<< 98:00:0A
[21:59:32][W][uartex:332]: [Read] Size error: 9850(2)
[21:59:32][D][uart_debug:114]: <<< 98:50:01:01:04:53:0A
[21:59:37][W][uartex:332]: [Read] Size error: 9801(2)
[21:59:38][D][uart_debug:114]: <<< 98:01:0A
[21:59:53][W][uartex:332]: [Read] Size error: 9818(2)
[21:59:53][D][uart_debug:114]: <<< 98:18:0A
[21:59:53][W][uartex:332]: [Read] Size error: 98(1)
[21:59:53][D][uart_debug:114]: <<< 98:54:01:0A
[21:59:53][W][uartex:332]: [Read] Size error: 9806(2)
[21:59:53][D][uart_debug:114]: <<< 98:06:0A
[21:59:53][W][uartex:332]: [Read] Size error: 98(1)
[21:59:53][D][uart_debug:114]: <<< 98:70:00:40:00:01:00:0A
```

The messages from the device always start with 0x98 and always end with 0x0A. The logs from uart_debug show that the correct messages are indeed received (i.e. 98:00:0A is status playing, 98:01:0A is status stop etc.). Commands instead start with 0x90 and end with 0x0A (i.e. 90:00:0A is status playing, 90:01:0A is status stop etc.), as said commands work fine.

Below is my current configuration espHome yaml configuration:

```
uart:
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: [0x0A]

# Example configuration entry
uartex:
  rx_timeout: 10ms
  tx_delay: 50ms
  tx_timeout: 500ms
  tx_retry_cnt: 3
  rx_header: [0x98]
  rx_footer: [0x0A]
  tx_header: [0x90]
  tx_footer: [0x0A]

media_player:
  - platform: uartex
    name: "CDP-CX225"
    state_playing:
      offset: 0
      data: [0x00]
    state_paused:
      offset: 0
      data: [0x02]
    state_idle:
      offset: 0
      data: [0x08]
    command_play:
      data: [0x00]
      ack: [0x00]
    command_stop:
      data: [0x01]
      ack: [0x01]
    command_pause:
      data: [0x02]
      ack: [0x02]

button:
  - platform: template
    name: "On"
    on_press:
      - uart.write: [0x90, 0x2E, 0x0A]
  - platform: template
    name: "Off"
    on_press:
      - uart.write: [0x90, 0x2F, 0x0A]
```

This device does not send any CRC.

How can I read data from this device?

JSa1987, 2025-05-31:
Hello,

I have tried to disable the debug and I still get the same error:

```
[08:56:25][W][uartex:332]: [Read] Size error: 982E(2)
[08:56:26][W][uartex:332]: [Read] Size error: 9861(2)
[08:56:33][W][uartex:332]: [Read] Size error: 9854(2)
[08:56:33][W][uartex:332]: [Read] Size error: 98(1)
[08:56:41][W][uartex:332]: [Read] Size error: 9858(2)
[08:56:45][W][uartex:332]: [Read] Size error: 9852(2)
[08:56:45][W][uartex:332]: [Read] Size error: 9808(2)
[08:56:48][D][media_player:074]: 'CDP-CX225' - Setting
[08:56:48][D][media_player:078]:   Command: PLAY
[08:56:48][D][uartex:208]: Write array-> 90000A(3)
[08:56:49][D][uartex:149]: Retry count: 1
[08:56:49][D][uartex:208]: Write array-> 90000A(3)
[08:56:49][W][uartex:332]: [Read] Size error: 9800(2)
[08:56:50][D][uartex:149]: Retry count: 2
[08:56:50][D][uartex:208]: Write array-> 90000A(3)
[08:56:50][W][uartex:332]: [Read] Size error: 9800(2)
[08:56:50][D][uartex:144]: Retry fail.
[08:56:50][W][uartex:352]: [Read] Tx Timeout error: (0)
[08:56:51][W][uartex:332]: [Read] Size error: 98(1)
[08:57:08][D][media_player:074]: 'CDP-CX225' - Setting
[08:57:08][D][media_player:078]:   Command: PAUSE
[08:57:08][D][uartex:208]: Write array-> 90020A(3)
[08:57:08][W][uartex:332]: [Read] Size error: 9802(2)
[08:57:09][D][uartex:149]: Retry count: 1
[08:57:09][D][uartex:208]: Write array-> 90020A(3)
[08:57:09][W][uartex:332]: [Read] Size error: 98(1)
[08:57:09][D][uartex:149]: Retry count: 2
[08:57:09][D][uartex:208]: Write array-> 90020A(3)
[08:57:09][W][uartex:332]: [Read] Size error: 9802(2)
[08:57:10][D][uartex:144]: Retry fail.
[08:57:10][W][uartex:352]: [Read] Tx Timeout error: (0)
[08:57:48][D][media_player:074]: 'CDP-CX225' - Setting
[08:57:48][D][media_player:078]:   Command: PLAY
[08:57:48][D][uartex:208]: Write array-> 90000A(3)
[08:57:48][W][uartex:332]: [Read] Size error: 9800(2)
[08:57:48][W][uartex:332]: [Read] Size error: 9800(2)
[08:57:49][D][uartex:149]: Retry count: 1
[08:57:49][D][uartex:208]: Write array-> 90000A(3)
[08:57:49][W][uartex:332]: [Read] Size error: 9800(2)
[08:57:49][D][uartex:149]: Retry count: 2
[08:57:49][D][uartex:208]: Write array-> 90000A(3)
[08:57:49][W][uartex:332]: [Read] Size error: 9800(2)
[08:57:50][D][uartex:144]: Retry fail.
[08:57:50][W][uartex:352]: [Read] Tx Timeout error: (0)
```

As reference below is my current configuration:

```
esphome:
  name: esp8266-test
  friendly_name: ESP8266-Test

esp8266:
  board: nodemcuv2

external_components:
  - source: github://eigger/espcomponents/releases/latest
    components: [ uartex ]
    refresh: 1days

# Enable logging
logger:
  level: DEBUG
  hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxx"

ota:
  - platform: esphome
    password: "xxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp8266-Test Fallback Hotspot"
    password: "xxxxxxxxx"

captive_portal:

uart:
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 115200
#  debug:
#    direction: BOTH
#   dummy_receiver: false
#    after:
#      delimiter: [0x0A]

# Example configuration entry
uartex:
  rx_timeout: 10ms
  tx_delay: 50ms
  tx_timeout: 500ms
  tx_retry_cnt: 3
  rx_header: [0x98]
  rx_footer: [0x0A]
  tx_header: [0x90]
  tx_footer: [0x0A]

media_player:
  - platform: uartex
    name: "CDP-CX225"
    state_playing:
      offset: 0
      data: [0x00]
    state_paused:
      offset: 0
      data: [0x02]
    state_idle:
      offset: 0
      data: [0x08]
    command_play:
      data: [0x00]
      ack: [0x00]
    command_stop:
      data: [0x01]
      ack: [0x01]
    command_pause:
      data: [0x02]
      ack: [0x02]

button:
  - platform: restart
    name: "ESP8266-Test AV Restart"
  - platform: template
    name: "On"
    on_press:
      - uart.write: [0x90, 0x2E, 0x0A]
  - platform: template
    name: "Off"
    on_press:
      - uart.write: [0x90, 0x2F, 0x0A]
```

JSa1987, 2025-05-31:
Never mind, I think I might have found the problem.
I have increased the time outs:

```
uartex:
  rx_timeout: 2000ms
  tx_delay: 50ms
  tx_timeout: 2000ms
  tx_retry_cnt: 3
  rx_header: [0x98]
  rx_footer: [0x0A]
  tx_header: [0x90]
  tx_footer: [0x0A]
```

And I'm not getting the error anymore:

```
[09:09:56][D][button:010]: 'Off' Pressed.
[09:10:06][D][button:010]: 'On' Pressed.
[09:10:13][D][media_player:074]: 'CDP-CX225' - Setting
[09:10:13][D][media_player:078]:   Command: PLAY
[09:10:13][D][uartex:208]: Write array-> 90000A(3)
[09:10:13][D][uartex:094]: Ack: 98000A(3), Gap Time: 82ms
[09:10:23][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[09:10:27][D][button:010]: 'Off' Pressed.
[09:10:30][D][button:010]: 'On' Pressed.
[09:10:49][D][media_player:074]: 'CDP-CX225' - Setting
[09:10:49][D][media_player:078]:   Command: PLAY
[09:10:49][D][uartex:208]: Write array-> 90000A(3)
[09:10:49][D][uartex:094]: Ack: 98000A(3), Gap Time: 79ms
[09:10:51][W][component:257]: Component uartex took a long time for an operation (70 ms).
[09:10:51][W][component:258]: Components should block for at most 30 ms.
[09:11:12][D][media_player:074]: 'CDP-CX225' - Setting
[09:11:12][D][media_player:078]:   Command: PAUSE
[09:11:12][D][uartex:208]: Write array-> 90020A(3)
[09:11:12][D][uartex:094]: Ack: 98020A(3), Gap Time: 79ms

```

Seems it takes about 80ms to get the ACK. I will run some more testing to make sure it all works well.

eigger, 2025-05-31:
It seems that the data isn't being received continuously. In that case, try using 100ms instead — 2000ms seems too large.
If the data is parsed correctly, there’s no major issue. However, if parsing fails midway, the loop could be blocked for 2000ms.

JSa1987, 2025-05-31:
Yes, after finding out that the timeout was the problem I had switched to 100ms.

I have re-enabled debugging and all works fine.

```
uart:
  tx_pin: GPIO01
  rx_pin: GPIO03
  baud_rate: 115200
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: [0x0A]

# Example configuration entry
uartex:
  rx_timeout: 100ms
  tx_delay: 50ms
  tx_timeout: 100ms
  tx_retry_cnt: 3
  rx_header: [0x98]
  rx_footer: [0x0A]
  tx_header: [0x90]
  tx_footer: [0x0A]
```

```
[18:50:26][D][button:010]: 'On' Pressed.
[18:50:26][D][uart_debug:114]: >>> 90:2E:0A
[18:50:26][D][uart_debug:114]: <<< 98:2E:0A
[18:50:26][D][uart_debug:114]: <<< 98:58:01:0A
[18:50:26][W][component:257]: Component uartex took a long time for an operation (51 ms).
[18:50:26][W][component:258]: Components should block for at most 30 ms.
[18:50:27][D][uart_debug:114]: <<< 98:58:01:0A
[18:50:27][D][uart_debug:114]: <<< 98:61:FE:00:0A
[18:50:27][W][component:257]: Component uartex took a long time for an operation (67 ms).
[18:50:27][W][component:258]: Components should block for at most 30 ms.
[18:50:32][D][uart_debug:114]: <<< 98:52:01:0A
[18:50:32][D][uart_debug:114]: <<< 98:08:0A
[18:50:38][D][media_player:074]: 'CDP-CX225' - Setting
[18:50:38][D][media_player:078]:   Command: PLAY
[18:50:38][D][uart_debug:114]: >>> 90:00:0A
[18:50:38][D][uartex:208]: Write array-> 90000A(3)
[18:50:38][D][uart_debug:114]: <<< 98:00:0A
[18:50:38][D][uartex:094]: Ack: 98000A(3), Gap Time: 78ms
[18:50:40][D][uart_debug:114]: <<< 98:50:01:01:04:53:0A
[18:50:40][W][component:257]: Component uartex took a long time for an operation (87 ms).
[18:50:40][W][component:258]: Components should block for at most 30 ms.
[18:50:51][D][media_player:074]: 'CDP-CX225' - Setting
[18:50:51][D][media_player:078]:   Command: PAUSE
[18:50:51][D][uart_debug:114]: >>> 90:02:0A
[18:50:51][D][uartex:208]: Write array-> 90020A(3)
[18:50:51][D][uart_debug:114]: <<< 98:02:0A
[18:50:51][D][uartex:094]: Ack: 98020A(3), Gap Time: 81ms
```

closed 2025-05-31 by JSa1987