eigger/espcomponents #147: GY-33 UART Color Sensor

LuckyLuzz, 2025-06-05:
Hey,
I've been trying to use your component to read data from a GY-33 color sensor. So far, I've managed to compile the YAML configuration, but my ESP32 keeps bootlooping and disconnecting after startup. Both Home Assistant (HA) and ESPHome are up to date.

I was hoping you could help me get UART communication working properly with this sensor.

I’ve attached:
- YAML Configuration: [YAML Color uart.txt](https://github.com/user-attachments/files/20608622/YAML.Color.uart.txt)
- UART Log Output: [output_2025-06-04_13-18-30.log.txt](https://github.com/user-attachments/files/20608637/output_2025-06-04_13-18-30.log.txt)

Here are additional references:
- GY-33 Sensor API Documentation: [GY-33 UART Protocol](https://github.com/QuirkyCort/micropython-gy33/tree/main/gy33-uart)
- Sensor Product Page: [GY-33 Color Sensor on Amazon](https://www.amazon.de/dp/B07DJ4N37B)

Any help in resolving this issue would be greatly appreciated!
Thanks in advance!

eigger, 2025-06-05:
Is the following format correct? Is there no checksum?
5A, 5A, 45, 03, 1F, 21, 1E,
head, head, type, length, red, green, blue

you can try like this

no checksum
```
uartex:
  id: uartex_gy33
  uart_id: uart_gy33
  rx_header: [0x5A, 0x5A]
  rx_length: 7

sensor:
  - platform: uartex
    state: [0x45]
    name: "GY-33 Red"
    unit_of_measurement: "%"
    state_number:
          offset: 2

  - platform: uartex
    state: [0x45]
    name: "GY-33 Green"
    unit_of_measurement: "%"
    state_number:
          offset: 3
```

or add checksum
```
uartex:
  id: uartex_gy33
  uart_id: uart_gy33
  rx_header: [0x5A, 0x5A]
  rx_length: 8
  rx_checksum: add

sensor:
  - platform: uartex
    state: [0x45]
    name: "GY-33 Red"
    unit_of_measurement: "%"
    state_number:
          offset: 2

  - platform: uartex
    state: [0x45]
    name: "GY-33 Green"
    unit_of_measurement: "%"
    state_number:
          offset: 3
```

LuckyLuzz, 2025-06-10:
Thanks, now it works! Thanks for the help. This might be a good sample for the Lib since there is not much to find via google.

[YAML Color uart.txt](https://github.com/user-attachments/files/20670371/YAML.Color.uart.txt)