eigger/espcomponents #173: UARTX: state_num precision not working anymore

JSa1987, 2025-07-20:
Since the update of ESPHome to rev. 2025.7 it seems that the `precision` configuration variable for `state_num` is not working anymore.

An ESPHome device that uses the  UARTX component, this has a couple of sensor defined with  `precision: 0`:

```
sensor:
  - platform: uartex
    name: "CD Loaded"
    id: cd_loaded
    icon: "mdi:disc"
    state: [0x58]
    state_number:
      offset: 1
      length: 1
      precision: 0
  - platform: uartex
    name: "CD Playing"
    id: cd_playing
    icon: "mdi:disc"
    state: [0x50]
    state_number:
      offset: 1
      length: 1
      precision: 0
```

These sensors used to show up in Home Assistant with now decimals (as intended). Since upgrading to ESPHome 2025.7.x these show up with one decimal as below.

 
[image]

eigger, 2025-07-21:
Could you please share your full configuration? 
The **precision** setting defines the precision of the raw data, but how many decimal places are shown in the UI is controlled by the **accuracy_decimals** option of the sensor

eigger, 2025-07-22:
This issue does not appear to originate from uartex, but rather from esphome or Home Assistant version 7.x. The same problem occurs even with simple template sensors.
```yaml
sensor:
  - platform: template
    name: "Template Sensor"
    lambda: |-
      return 4; #4.0 same
    update_interval: 5s
```

[image]

closed 2025-07-23 by eigger